MIPS/CI20_linux #16: ext4load 3.18 kernel boothang

ZubairLK, 2015-01-28:
uboot nand. kernel on nand. boots
uboot mmc. kernel on mmc. Hangs at https://gist.github.com/Etn40ff/e485226c35a250078a46

uboot mmc. tftp loading the kernel. boots.

Very strange.

Disabling mmc in kernel boots too.

ZubairLK, 2015-01-28:
ext4load the kernel
then tftp the kernel.

hangs again.

suggests uboot leaves mmc in a bad state.

perhaps theres a wait loop somewhere that needs to be timed out

http://pastebin.com/tgN9pJHx

paulburton, 2015-01-28:
I don't consider this a U-boot bug. The kernel should really be able to deal with whatever state the hardware is in at boot time, and given that older kernels can I'd consider this a regression in the ci20-v3.18 branch.

Etn40ff, 2015-01-28:
3.16 can't either. Not sure about 3.0.8

ZubairLK, 2015-01-29:
3.0.8 works..

pcercuei, 2015-01-29:
I have this problem with 3.18 as well. But I never had problems with 3.16.

ZubairLK, 2015-01-29:
I've traced it and fixed this.

irq thread is attached before mmc reset.

In one particular case, mmc init from uboot and brcm sdio wifi mmc1 enabled, this hangs the ci20 as the irq loops.

Kernel should be ok with whatever state uboot leaves devices in.

Will push out the fix in a while.

ZubairLK, 2015-01-29:
See this https://github.com/MIPS/CI20_linux/commit/26899e632969cc2533b689c1029e4ec59fcef349

It boots ok from mmc for me now using ext4load.

closed 2015-02-16 by ZubairLK