MIPS/CI20_linux #20: Camera not working with Android 4.4.2 and Kernel Version 3.0.8

quantumaardvark, 2015-02-10:
I'd be grateful for any help/guidance.

I've just got my C120 and have loaded installed the android4.4.2_20141126 image. 

The kernel is 3.0.8-12479-g505af31.

I presume the kernel is also used with Debian 8 as there is a note against that stating Mplayer is removed and PowerVR core driver has not yet been ported. Does anyone know if this is the reason the camera (the same one cited in the hardware pages) does not work? Before I installed Android I ran the Debian demos and the PowerVR ones worked fine but the desktop camera application reported Mplayer was missing.

I'd be very grateful if anyone know the status of these two drivers.

ehabhussein, 2015-02-10:
To get camera support on the CI20 for linux you need to download http://mipscreator.imgtec.com/CI20/images/default_NAND/Debian7_20150115/debian7_2015_01_15.img and follow the instructions on this page http://elinux.org/CI20_Dev_Zone#NAND_Flashing_SD_card.

quantumaardvark, 2015-02-10:
Ehab, many thanks for this. I see that this is a Debian 7 image, and hence the drivers are present. Thanks again.

closed 2015-02-10 by quantumaardvark