MIPS/CI20_linux #23: kernel cannot be built directly on the ci20

mcmechan, 2015-04-12:
Building on the ci20 requires several packages to be installed in the latest debian image

`apt-get install bc u-boot-tools`

and then

`make -j3 ci20_defconfig uImage`

When compiled on my ext4 uSD card it takes about a hour

Personally I don't like to do things like compiles on non-removable media.

Also libncurses5-dev is required for make menuconfig

mcmechan, 2015-04-13:
Also if you want access to the NAND you will need to install mtd-utils for the ubiattach command unless you are using as the boot device.

jcowgill, 2015-04-14:
Ignoring #22, I don't see the bug here - having to install extra build dependencies is pretty common.

ZubairLK, 2015-04-14:
#22 or #23 ? #22 is different.

And yes. These dependencies are needed on x86 systems as well before compiling the kernel

It is certainly not a kernel bug. Having them on the default images on the ci20 is a nice to have perhaps. For some use cases. But it will bloat the image further.

jcowgill, 2015-04-14:
> #22 or #23 ? #22 is different.

You can't build the kernel directly on the CI20 without #22 being fixed first, even if you have all the build dependencies.

mcmechan, 2015-04-14:
Well you can build the kernel it just takes some work arounds, currently I have 23 hours of uptime on a ci20-v3.18.3 kernel I built on my ci20 with a SD card and a patched mkimage.  The missing dependancies were a fairly minor issue, likely best fixed by updating the documentation #22 was a pain to find.
I first worked around it by manually running the mkuboot.sh script that the kernel uses with the parameters fixed up by hand that kernel would work.

kernel version:

/bin/bash ./scripts/mkuboot.sh -A mips -O linux -C gzip -T kernel -a 0xffffffff80010000 -e 0x805bdc10 -n Linux-3.18.3+ -d arch/mips/boot/vmlinux.bin.gz arch/mips/boot/uImage.gz

and then I rewrote it so it would work and ran it by hand:

/bin/bash ./scripts/mkuboot.sh -A mips -O linux -C gzip -T kernel -a 0x80010000 -e 0x805bdc10 -n Linux-3.18.3+ -d arch/mips/boot/vmlinux.bin.gz arch/mips/boot/uImage.gz

Then I tried to update the kernel but it appears that it is using the 0xffffffff80010000 as a magic flag so if you take off the high 32bits it is not able to build a uImage, so I just patched mkimage in u-boot. see #22 for the sed script I made, I have not figured out how to include a patch yet.

ZubairLK, 2015-04-28:
I think this is an mkimage issue depending on 32/64bit host..

See the discussion here as well

http://patchwork.linux-mips.org/patch/4447/

wcohen, 2016-02-22:
A patch submitted to u-boot mailing list (https://patchwork.ozlabs.org/patch/583426/) has enabled self-hosted build of the linux-3.18.3 kernel checked out from https://github.com/MIPS/CI20_linux.git .

ZubairLK, 2016-02-29:
Thank-you very much @wcohen

closed 2016-02-29 by ZubairLK