MIPS/CI20_linux #24: kernel is missing files required by ci20_defconfig

mcmechan, 2015-04-12:
Tthe ci20_defconfig has

`CONFIG_EXTRA_FIRMWARE="brcm/brcmfmac4330-sdio.bin brcm/brcmfmac4330-sdio.txt"`

neither file shows up when cloned and no directions for downloading them from elsewhere are on the wiki page.
This causes the compile to fail with unresolved dependancies.

mcmechan, 2015-04-13:
I found the files in the debian distribution at
ci20:/lib/firmware/iw8103/fw_bcm4330b2.bin -> CI20_linux/firmware/brcm/brcmfmac4330-sdio.bin
ci20:/lib/firmware/iw8103/nv_4330b2.txt  -> CI20_linux/firmware/brcm/brcmfmac4330-sdio.txt
This should be documented somewhere people can find it.
The renameing of the files threw me for a couple days and having it under iw8103 did not help.

ZubairLK, 2015-04-14:
That is correct. I thought I answered the same question on irc. Should have updated the pull request instead.

There are some legalities surrounding the shipping of binaries alongside GPL sources. [1] 

So we can't put them within the kernel repo.

The ci20_defconfig for 3.18 loads the firmware inside the kernel file. It should ideally not do that and load the firmware from user-space the way 3.0.8 does it.

But you are correct. Until user-space, this should definitely be wiki'd somewhere on elinux.
I put it here http://elinux.org/CI20_upstream#WiFi_firmware

[1] https://lwn.net/Articles/130696/

mcmechan, 2015-04-14:
Thank you, I just submitted this because I could not find a version that explained where to get them, and here people might be able to find it with google.

closed 2015-04-24 by mcmechan