MIPS/CI20_linux #25: Kernel v3.18.3+ spins 2nd cpu

mcmechan, 2015-04-24:
When I succeeded in building v3.18.3+ on the ci20 it booted fine but the 2nd CPU sits at 100%
top was reporting rpcbind as 100% (or 50% of total CPU use) when I killed rpcbind the CPU load still did not drop but it was not in any process shown by top...
I was using top with "1" and "I" to show individual CPU stats in the totalled vs non-totalled mode

jcowgill, 2015-04-25:
`cat /proc/stat` on my ci20 gives:
```
#    user   nice   system idle      io    irq       softirq
cpu  890971 212506 266051 254229651 37111 254761780 168504 0 0 0
cpu0 490938  91070 266051 254229651 37111         0 168504 0 0 0
cpu1 400033 121436      0         0     0 254761780      0 0 0 0
```
I can also put load onto the second CPU and it seems to work - so my guess is it's just the statistics in `/proc/stat` that are wrong.

mcmechan, 2015-04-26:
When I check on my otherwise idle system I seem to have pretty high numbers
now it is possible that this is because I am using a kernel compiled on the ci20 itself.
But mostly if you can get it to compile natively it is pretty correct.

```
root@ci20:~# cat /proc/stat 
cpu  5926443 0 167195 10000582 534 4513926 215 0 0 0
cpu0 135934 0 167195 10000582 534 0 215 0 0 0
cpu1 5790509 0 0 0 0 4513926 0 0 0 0
intr 52449682 0 0 20892517 10664052 0 0 0 0 0 0 0 1 0 0 0 0 0 0 60736 0 0 0 0 0 0 0 0 0 0 2 0 0 0 10304438 10304459 0 0 0 0 0 0 0 28 0 115639 106483 0 0 0 0 0 0 0 0 0 0 0 10 0 54 0 0 0 0 33 0 0 0 104 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 1126 0 0
ctxt 3294471
btime 1429988114
processes 2762
procs_running 2
procs_blocked 0
softirq 32649405 1 20608894 673 8403 0 0 60781 11097108 0 873545
```
and still top seems to be saying that I am using a lot of user and high
```
top - 23:33:49 up 1 day,  4:38,  1 user,  load average: 1.01, 1.03, 1.05
Tasks:  75 total,   2 running,  73 sleeping,   0 stopped,   0 zombie
%Cpu(s): 26.2 us,  0.2 sy,  0.0 ni, 49.7 id,  0.0 wa, 24.0 hi,  0.0 si,  0.0 st
KiB Mem:   1024020 total,   106420 used,   917600 free,     8540 buffers
KiB Swap:  2095100 total,        0 used,  2095100 free,    68364 cached

  PID USER      PR  NI  VIRT  RES  SHR S  %CPU %MEM    TIME+  COMMAND                                     
 1779 root      20   0  3004 1888 1712 R 100.0  0.2   1718:14 rpcbind                                     
 2762 root      20   0  3740 2336 1940 R   0.7  0.2   0:00.04 top                                         
    1 root      20   0  2636 1576 1472 S   0.0  0.2   0:01.44 init                                        
```
I should also note that the CPU is not hot and that this is on the Debian image on a SD card with a kernel compiled under the Debian on the SD card, and the two files renamed as per the now closed issue #24 and the u-boot: mkimage.c patched for 64 integers per issue #22

ZubairLK, 2015-04-28:
what is your cmdline. I noticed this rpcbind cpu usage would go away with devtmpfs.mount=1 in the cmdline. I thought I had enabled it in the defconfig..

ZubairLK, 2015-04-28:
Please note. Due to the way interrupts are wired inside the jz4780.

One cpu will always have to handle the interrupts. That might mess up stats..

ZubairLK, 2015-06-24:
Should be solved by 58aab7326ee6e1e637bd25a10b177990c16753bc as it enables CONFIG_DEVTMPFS_MOUNT=y in defconfig

closed 2015-06-24 by ZubairLK