MIPS/CI20_linux #27: Unable to build kernel module z4780_drv  on v3.18

axjslack, 2015-06-12:
I try to build the kernel (3.18 commit: 8ef5089c1baf62db99f0855d49bd2fb9b89be685) but it failed.

First of all the default ci20_defconfig fail on building firmware of wireless sdio card. Anyway I removed for the moment the driver, and I'll look it later. 

It failed during building of z4780_drv.o (I try to build it as module and not built-in) I add a log at the end of the message.

I'm using this compiler:

--- Compiler ---
mips-linux-gnu-gcc -v
Using built-in specs.
COLLECT_GCC=mips-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/home/alexjan/build/mips_qemu/output/host/opt/ext-toolchain/bin/../libexec/gcc/mips-linux-gnu/4.9.1/lto-wrapper
Target: mips-linux-gnu
Configured with: /scratch/cmoore/final-2014.11-linux-lite/src/gcc-4.9-2014.11/configure --build=i686-pc-linux-gnu --host=i686-pc-linux-gnu --target=mips-linux-gnu --enable-threads --disable-libmudflap --disable-libssp --disable-libstdcxx-pch --with-arch-32=mips32r2 --with-arch-64=mips64r2 --with-float=hard --with-mips-plt --enable-extra-sgxxlite-multilibs --with-gnu-as --with-gnu-ld --with-specs='%{save-temps: -fverbose-asm} -D__CS_SOURCERYGXX_MAJ__=2014 -D__CS_SOURCERYGXX_MIN__=11 -D__CS_SOURCERYGXX_REV__=22' --enable-languages=c,c++ --enable-shared --enable-lto --enable-symvers=gnu --enable-__cxa_atexit --with-glibc-version=2.20 --with-pkgversion='Sourcery CodeBench Lite 2014.11-22' --with-bugurl=https://sourcery.mentor.com/GNUToolchain/ --disable-nls --prefix=/opt/codesourcery --with-sysroot=/opt/codesourcery/mips-linux-gnu/libc --with-build-sysroot=/scratch/cmoore/final-2014.11-linux-lite/install/opt/codesourcery/mips-linux-gnu/libc --with-gmp=/scratch/cmoore/final-2014.11-linux-lite/obj/pkg-2014.11-22-mips-linux-gnu/mips-2014.11-22-mips-linux-gnu.extras/host-libs-i686-pc-linux-gnu/usr --with-mpfr=/scratch/cmoore/final-2014.11-linux-lite/obj/pkg-2014.11-22-mips-linux-gnu/mips-2014.11-22-mips-linux-gnu.extras/host-libs-i686-pc-linux-gnu/usr --with-mpc=/scratch/cmoore/final-2014.11-linux-lite/obj/pkg-2014.11-22-mips-linux-gnu/mips-2014.11-22-mips-linux-gnu.extras/host-libs-i686-pc-linux-gnu/usr --with-isl=/scratch/cmoore/final-2014.11-linux-lite/obj/pkg-2014.11-22-mips-linux-gnu/mips-2014.11-22-mips-linux-gnu.extras/host-libs-i686-pc-linux-gnu/usr --with-cloog=/scratch/cmoore/final-2014.11-linux-lite/obj/pkg-2014.11-22-mips-linux-gnu/mips-2014.11-22-mips-linux-gnu.extras/host-libs-i686-pc-linux-gnu/usr --with-libelf=/scratch/cmoore/final-2014.11-linux-lite/obj/pkg-2014.11-22-mips-linux-gnu/mips-2014.11-22-mips-linux-gnu.extras/host-libs-i686-pc-linux-gnu/usr --enable-libgomp --disable-libitm --enable-libatomic --disable-libssp --enable-poison-system-directories --with-build-time-tools=/scratch/cmoore/final-2014.11-linux-lite/install/opt/codesourcery/mips-linux-gnu/bin --with-build-time-tools=/scratch/cmoore/final-2014.11-linux-lite/install/opt/codesourcery/mips-linux-gnu/bin SED=sed
Thread model: posix
gcc version 4.9.1 (Sourcery CodeBench Lite 2014.11-22) 

--- End Compiler ---
--- Log ---
  CC [M]  drivers/gpu/drm/jz4780/jz4780_drv.o
In file included from include/linux/printk.h:5:0,
                 from include/linux/kernel.h:13,
                 from include/linux/clk.h:16,
                 from drivers/gpu/drm/jz4780/jz4780_drv.h:21,
                 from drivers/gpu/drm/jz4780/jz4780_drv.c:20:
include/linux/init.h:328:27: error: redefinition of '__inittest'
  static inline initcall_t __inittest(void)  \
                           ^
drivers/gpu/drm/jz4780/jz4780_drv.c:374:1: note: in expansion of macro 'module_init'
 module_init(jz4780_drm_init);
 ^
include/linux/init.h:328:27: note: previous definition of '__inittest' was here
  static inline initcall_t __inittest(void)  \
                           ^
include/linux/init.h:320:28: note: in expansion of macro 'module_init'
 #define late_initcall(fn)  module_init(fn)
                            ^
drivers/gpu/drm/jz4780/jz4780_drv.c:373:1: note: in expansion of macro 'late_initcall'
 late_initcall(jz4780_drm_init);
 ^
include/linux/init.h:330:6: error: redefinition of 'init_module'
  int init_module(void) __attribute__((alias(#initfn)));
      ^
drivers/gpu/drm/jz4780/jz4780_drv.c:374:1: note: in expansion of macro 'module_init'
 module_init(jz4780_drm_init);
 ^
include/linux/init.h:330:6: note: previous definition of 'init_module' was here
  int init_module(void) __attribute__((alias(#initfn)));
      ^
include/linux/init.h:320:28: note: in expansion of macro 'module_init'
 #define late_initcall(fn)  module_init(fn)
                            ^
drivers/gpu/drm/jz4780/jz4780_drv.c:373:1: note: in expansion of macro 'late_initcall'
 late_initcall(jz4780_drm_init);
 ^
make[4]: *** [drivers/gpu/drm/jz4780/jz4780_drv.o] Error 1
make[3]: *** [drivers/gpu/drm/jz4780] Error 2
make[2]: *** [drivers/gpu/drm] Error 2
make[1]: *** [drivers/gpu] Error 2
make: *** [drivers] Error 2
--- End Log ---

ZubairLK, 2015-06-12:
Is there a particular reason for selecting the driver as a module? 

I haven't tried doing that. So dunno.

IIRC, the driver needs a particular initialization sequence. Otherwise driver probes fail.

drm needs i2c. Because i2c is used by edid. Which is used by hdmi. which is needed by drm.
:)

axjslack, 2015-06-12:
I'm testing various configuration, and one of this is to reduce the amount of driver built-in, in order to reduce kernel image size (in particular for modules that are not strictly needed in boot stage).

 Anyway, if the driver is build as built-in it builds. If driver is built as a module it fail.

jcowgill, 2015-06-12:
In https://github.com/MIPS/CI20_linux/blob/ci20-v3.18/drivers/gpu/drm/jz4780/jz4780_drv.c#L373

One of these lines needs to be dropped:
```c
late_initcall(jz4780_drm_init);
module_init(jz4780_drm_init);
```

Modules can only have one init function.

HarveyHunt, 2015-06-13:
In the ```Wifi Firmware``` section of this page is a fix for the wifi drivers compile error:

http://www.elinux.org/CI20_upstream

axjslack, 2015-06-15:
For the moment I will build as built-in. I think that this is the temporary solution. Thanks to all for the answers.

jcowgill, 2015-06-16:
That's a workaround, but the issue itself is still here - you should be able to compile it as a module.

jhdulaney, 2015-10-13:
In addition, you're linking with dwc_hdmi.  Rather than link with it, how about depend on it?